mongoose-os-apps/shelly-homekit #1607: Shelly 1 Plus reverts back to switch from light

jdesutter, 2025-01-06:
Converted from stock to HomeKit and setup switch as a light in HomeKit and it stays for a few days and then reverts back to switch  which then messes up scenes and disappears from control panel on iOS 18 on my phone and has to be re-setup. When it reverts back HomeKit makes a bridge with 2 devices for the same shelly 1 plus--one that works and one that doesn't respond. Sometimes there is a working version of the switch without any bridge. I have to wipe them out and start over. i did notice on the switch screen an input #2 that is disabled. Don't understand what or why its there. The system acts as though there are 2 devices. Appreciate any help you can give me. Thank you, Joe

markirb, 2025-01-06:
HI, you already noticed this bug is described here:
https://github.com/mongoose-os-apps/shelly-homekit/issues/1417
you can also find a firmware that fixes it there.

closed 2025-01-06 by markirb

jdesutter, 2025-01-06:
Hi Markus,Yes I thought that might be the fix, unfortunately I’m an amateur and don’t know how to flash it. I downloaded it and unzipped it and tried to upload it on the firmware tap but it didn’t like any of those files. Perhaps you could send me some instructions and I would greatly appreciate it. Thank you,JoeSent from my iPhone