mongoose-os-apps/shelly-homekit #1613: When reverting to factory default ShellyMini1PMG3 dead

tom666-max, 2025-01-13:
I did a revert with the Shelly HomeKit 2.13 web interface. The revert stopped at 97%. Since then the Shelly has been dead. A factory reset with the s button 10s after restart is no longer possible, nor is 5x on off after the restart. The LED no longer lights up either. Does anyone have an idea?

tom666-max, 2025-01-15:
Yes, I am

Show:

https://github.com/user-attachments/assets/3466e657-41bf-489b-b547-c79505bf9e0a

tom666-max, 2025-01-15:
Thank you for the feedback. As far as I can tell, I need to solder on a serial interface and then somehow run the attached config file. Is that correct and how do I do it exactly? Is there an easier way?

tom666-max, 2025-01-16:
would it have been safer if I had done a reset first so that it was already in AP mode and then reverted for future actions?

markirb, 2025-01-16:
Yes solder an interface. Then you can run the mos flash tool. 

If you do that it would be most interesting for me what the boot console has to say before you flash.

Factory reset before reverting would be the only tip I can give yes.

tom666-max, 2025-01-16:
Thank you very much for the information
Unfortunately I can't find the interface anymore. I used it for firescript a long time ago. And it's not worth ordering one separately as I would probably only use it once, so I might as well buy a new shelly. In future I will always do a factory reset before reverting. If you are in Switzerland, I could send you the shelly so you can see what the problem is.

markirb, 2025-01-18:
I'm in Germany, so if you want to then yes I would look at what the problem is.

tom666-max, 2025-01-18:
OK, next time I'm in Germany I can send it to you. What's your address?

markirb, 2025-01-20:
Can you please send an email to the address in my GitHub profile? Then I will write you my address.

markirb, 2025-01-23:
@petealves do you mean it has no wifi or do you just mean the relay does not work? it is not clear from your description. if only the relay is not working as it sounds then a hard power cut should be enough.

petealves, 2025-01-23:
@markirb i am sorry for not explaining it correctly.
It has no wifi at all. Mini 1 Gen 3 also has a status LED, which also won't turn on. What i meant to say about the relay is that when you have the shelly ready to pair and you feed power to it, a little "click" of the relay can be heard. That click is not heard.
I have already tried to reset through the button (10 secs), nothing, tried with a switch and flipping it, also won't work. It won't show in bluetooth nor AP is showing

markirb, 2025-01-23:
I see. I will look at the other hardware as soon as I have it to know the reason. 

But I fear that something breaks the revert currently.

Only way to know is the serial interface.

markirb, 2025-01-23:
Which stock firmware version did you use to flash originally? This might be of interest

petealves, 2025-01-23:
I tried the revert via the WebUI of the Mongoose. I didn't pay attention to be honest. I have 20+ Shellys with Mongoose all working correctly and they always reverted OK when i needed via the Web UI, so, i just didn't pay attention to the version :/

markirb, 2025-01-30:
So, thanks @tom666-max I got the Shelly in the mail today and looked at the serial output.

The revert to stock was successful however the Stock-Fw boots and looks for the partition table where there is none. It follows a boot/crash loop that is non recoverable.

I have to look much deeper but the preliminary conclusion is:
As the bootloader does not seem to revert even after several unsuccessful boots (it should to my understanding), this will not be recoverable without soldering an interface and flashing via this interface.

Shelly seems to keep moving the partition table around in recent versions. As precaution their partition table contains redundant data, which HomeKit firmware currently does not have. So I have to look deeper but it should definitely be possible in future to avoid this situation.

tom666-max, 2025-01-31:
Hello Markus, what exactly happens when you revert? Is the FW that was there before loaded or is a new version loaded? Is the partition table repartitioned? And is the bootloader also replaced?
Regards, Thomas

markirb, 2025-01-31:
Bootloader is not flashed.
Partition table normally also not according to my knowledge, at least in previous versions. however it seems to do it in recent ones.

Did you happen to have 1.5beta installed before?

tom666-max, 2025-01-31:
no, it was 1.3.x or 1.4.x.

damianfryc, 2025-02-02:
Same here. Shelly 1 Gen 3 revert to stock fw from gui and device is dead. Any solution?

markirb, 2025-02-02:
As said above:

As the bootloader does not seem to revert even after several unsuccessful boots (it should to my understanding), **this will not be recoverable without soldering an interface and flashing via this interface.**

Investigation of how we avoid this will follow...

damianfryc, 2025-02-03:
> As said above:
> 
> As the bootloader does not seem to revert even after several unsuccessful boots (it should to my understanding), **this will not be recoverable without soldering an interface and flashing via this interface.**
> 
> Investigation of why this happens (it does not happen on mine) and how we avoid this will follow...

Thank you. I see post above with pinout on Shelly but do I need any additional device or just I have to soldier this to DB-9(RS232) connector?

markirb, 2025-02-03:
Check here for what you need.

https://tasmota.github.io/docs/Getting-Started/

I was playing around with this today:
Upgrade and Downgrade Pathes work smoothly with 1.4.4 and even 1.5b.
Maybe there was some version before that made an error here...

I do understand the problem but I do not know why this happened.
Newest Shelly-OS can and do upgrade bootloader as well as partition tables. mongoose does not have/do that.

markirb, 2025-02-06:
So I have some thought.

The update Chain is:

1. Original Stock FW Version
2. Shelly Homekit Version
3. Revert Stock FW Version

Most interesting for this error case is 1. Original Stock version.
I fear that this problem only arises when 1. was smaller (maybe also equal) than 1.3.3.

Can anyone of you remember where they came from? I will play around with that when I find time.

tom666-max, 2025-02-07:
I'm pretty sure I went from stock 1.3.3 to Shelly HomeKit 2.12.2 and then used the webgui to do all the updates up to 2.13.3 when they were released.

AlePitz, 2025-02-08:
Hi, I had the same issue a few days ago, and now my Shelly is dead. I need to revert to stock firmware 10-12 other Shellys, how can I do that without killing my Shelly? Does OTA work or is it dangerous to try? Thanks

markirb, 2025-02-08:
It is only dangerous for Minis so far as all others Plus Models can only come from 1.4 due to the changed partition layout. 
Or they automatically downgrade back to correct version due to changed urls on our side.

When on Minis you should manually revert to the one you came from. 
1.3.3 versions you can find here:
https://rojer.me/files/shelly/stock/1.3.3/

tom666-max, 2025-02-08:
It's great that there is a safe option. Unfortunately I don't know which version was on every Shelly. Is it possible to find out or does 1.3.3 work for those who had 1.4 on them?

markirb, 2025-02-08:
Unfortunately there is currently none.

Do you have so many Minis? Why do you want to revert them all in the first place?

You could read back the Partitions via UART and check, but you need to solder an interface then.

It could be that a Downgrade to Mini 1.3 would be always safe, if Shelly does not zero out the Partition Table when moving it. I cannot say if this is the case.

Other option is to implement somehow a read back of firmware on the other partition, but that's quite some work

tom666-max, 2025-02-09:
The Shelly HomeKit is a great thing.

Yes I have many 1PM Mini Gen3.

I want to pair them with Apple Home and Home Assistant at the same time.

This works with the MINIR4M without problems.

Home Assistant also works easily with the "shelly homekit" and 1PM Mini Gen3 just not at the same time as Apple Home.

HomeKit settings says 16max, so that should be possible.

To test other variants, I then did the reverse, which made a Shelly unuseable. I then refrained from further tests so as not to damage other 

[image]

[image]

devices.

AlePitz, 2025-02-09:
I just tried reverting to 1.3.3 and it did work for every Shelly I used. They didn't come from 1.3.3, it was an older version. Hope that helps

alx080, 2025-05-04:
Can you tell me if I can manually and safely revert my Shelly 1 Mini G3—which originally came with firmware 1.3.3, was then updated to 1.5.1 as described here, and subsequently flashed with the Shelly HomeKit firmware—back to version 1.3.3, or might that process fail? As far as I can see, only version 1.3.3 is available for selection.

tom666-max, 2025-05-05:
> Can you tell me if I can manually and safely revert my Shelly 1 Mini G3—which originally came with firmware 1.3.3, was then updated to 1.5.1 as described here, and subsequently flashed with the [...]

For me, it once didn't work with the "1PM Mini Gen3." I originally updated from 1.4 to Shelly Homekit 2.13. When I reverted from 2.13 to 1.3.3, Shelly became unusable.

Afterward, I always reset Shelly Homekit before reverting, and it always worked, regardless of whether it was originally 1.4 or 1.5.1.

ppawiggers, 2025-08-26:
Just tried to revert my MiniG3 to 1.3.3 from Shelly Homekit 2.14. It seemed to work (it rebooted after the firmware update) but when I open the web interface it hangs at "Connecting to Shelly".

Tried to revert back to Homekit with `http:/[ip]/ota?url=http://shelly.rojer.cloud/update`. That returns `Trying to update from http://shelly.rojer.cloud/update` but it doesn't do anything.

It seems my Shelly is unusable now.

Reason to revert to stock: it now has native support for Matter.