mongoose-os-apps/shelly-homekit #1616: rgbwPM in white mode 4 channels doesn't work

stahlbox, 2025-01-20:
Hello,

I have the following problem with the Shelly rgbwPM:
I want to use it in white mode (4 channels) with 4 LED strips and 4 buttons (one for each channel) . The test installation was carried out with 2 LEDs on 2 channels with 2 buttons. I want to use one button for each LED and the Homekit control. As long as I use button 1 and only channel 1 in the Homekit, LED 1 works as it should. As soon as I use channel 2 via the button or Homekit, both channels switch at the same time. If channel 1 is off and I activate channel 2, channel 1 is also activated automatically. I can then only control both channels. Even if I then control channel 1, both channels are controlled.

The wiring is correct according to Shelly specifications in white mode and the circuit works perfectly with the Shelly software.

Input mode = momentary
initial state = last

Same thing after a hard reset.

Does anyone have any ideas?

markirb, 2025-01-20:
Currently not. 

Can you please send logs when this happens?

stahlbox, 2025-01-20:
It seems the channel 1 works fine by physical and digital switch. As soon as I activate channel 2 all channels are activated by all physical and digital switches..

stahlbox, 2025-01-20:
Log by activate channel 1: (I have deleted the transition parts) 

18682805 shelly_debug.cpp:231    No log file, sending new entries
24581770 shelly_main.cpp:492     Up 24.56, HAP 0/1/16 ns 1, RAM: 204240/193860; st 0; 11.1: sta: off, b: 100, h: 0, sa: 100, ct: 400; 11.2: sta: off, b: 100, h: 0, sa: 100, ct: 400; 11.3: sta: off, b: 100, h: 0, sa: 100, ct: 400; 11.4: sta: off, b: 54, h: 0, sa: 100, ct: 400
32581739 shelly_main.cpp:492     Up 32.56, HAP 0/1/16 ns 1, RAM: 204240/193860; st 0; 11.1: sta: off, b: 100, h: 0, sa: 100, ct: 400; 11.2: sta: off, b: 100, h: 0, sa: 100, ct: 400; 11.3: sta: off, b: 100, h: 0, sa: 100, ct: 400; 11.4: sta: off, b: 54, h: 0, sa: 100, ct: 400
37669769 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.30:57702
37974361 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.30:57702
38998058 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.30:57702
40580954 shelly_main.cpp:492     Up 40.56, HAP 0/1/16 ns 1, RAM: 204316/193860; st 0; 11.1: sta: off, b: 100, h: 0, sa: 100, ct: 400; 11.2: sta: off, b: 100, h: 0, sa: 100, ct: 400; 11.3: sta: off, b: 100, h: 0, sa: 100, ct: 400; 11.4: sta: off, b: 54, h: 0, sa: 100, ct: 400
48581496 shelly_main.cpp:492     Up 48.56, HAP 0/1/16 ns 1, RAM: 204316/193860; st 0; 11.1: sta: off, b: 100, h: 0, sa: 100, ct: 400; 11.2: sta: off, b: 100, h: 0, sa: 100, ct: 400; 11.3: sta: off, b: 100, h: 0, sa: 100, ct: 400; 11.4: sta: off, b: 54, h: 0, sa: 100, ct: 400
49951416 shelly_input.cpp:77     Input 1: change (state 0)
50170422 shelly_input.cpp:77     Input 1: change (state 1)
50175905 shelly_hap_light_bu:198 State changed (ext_mom): off => on
50183468 shelly_light_bulb_c:102 Starting transition: [w=0.00] -> [w=1.00], 2000 ms
50187968 shelly_white_control:38 Output 1: 0.00 => 1.00
50199781 shelly_output.cpp:83    Duty: 0.009
50205923 shelly_output.cpp:96    Output 1: 0.009436 (transition)
50210708 shelly_output.cpp:83    Duty: 0.015

52167712 shelly_output.cpp:96    Output 1: 0.991171 (transition)
52173093 shelly_output.cpp:83    Duty: 0.997
52178471 shelly_output.cpp:96    Output 1: 0.996566 (transition)
52186795 shelly_light_bulb_co:56 Transition finished, end state: [w=1.00]
52190790 shelly_output.cpp:83    Duty: 1.000
52196440 shelly_output.cpp:96    Output 1: 1.000000 (transition)
52628865 shelly_input.cpp:77     Input 1: change (state 0)
52852351 shelly_input.cpp:77     Input 1: change (state 1)
52857830 shelly_hap_light_bu:198 State changed (ext_mom): on => off
52865343 shelly_light_bulb_c:102 Starting transition: [w=1.00] -> [w=0.00], 2000 ms
52869843 shelly_white_control:38 Output 1: 1.00 => 0.00
52879042 shelly_output.cpp:83    Duty: 0.991
52884460 shelly_output.cpp:96    Output 1: 0.991436 (transition)
52890532 shelly_output.cpp:83    Duty: 0.986

54849330 shelly_output.cpp:96    Output 1: 0.010104 (transition)
54854187 shelly_output.cpp:83    Duty: 0.004
54859569 shelly_output.cpp:96    Output 1: 0.003810 (transition)
54868304 shelly_light_bulb_co:56 Transition finished, end state: [w=0.00]
54872147 shelly_output.cpp:83    Duty: 0.000
56651053 mgos_sys_config.c:323   Saved to conf9.json
56674917 shelly_main.cpp:492     Up 56.65, HAP 0/1/16 ns 1, RAM: 204116/193860; st 0; 11.1: sta: off, b: 100, h: 0, sa: 100, ct: 400; 11.2: sta: off, b: 100, h: 0, sa: 100, ct: 400; 11.3: sta: off, b: 100, h: 0, sa: 100, ct: 400; 11.4: sta: off, b: 54, h: 0, sa: 100, ct: 400
64580956 shelly_main.cpp:492     Up 64.56, HAP 0/1/16 ns 1, RAM: 204320/193860; st 0; 11.1: sta: off, b: 100, h: 0, sa: 100, ct: 400; 11.2: sta: off, b: 100, h: 0, sa: 100, ct: 400; 11.3: sta: off, b: 100, h: 0, sa: 100, ct: 400; 11.4: sta: off, b: 54, h: 0, sa: 100, ct: 400
72580997 shelly_main.cpp:492     Up 72.56, HAP 0/1/16 ns 1, RAM: 204320/193860; st 0; 11.1: sta: off, b: 100, h: 0, sa: 100, ct: 400; 11.2: sta: off, b: 100, h: 0, sa: 100, ct: 400; 11.3: sta: off, b: 100, h: 0, sa: 100, ct: 400; 11.4: sta: off, b: 54, h: 0, sa: 100, ct: 400
80580972 shelly_main.cpp:492     Up 80.56, HAP 0/1/16 ns 1, RAM: 204320/193860; st 0; 11.1: sta: off, b: 100, h: 0, sa: 100, ct: 400; 11.2: sta: off, b: 100, h: 0, sa: 100, ct: 400; 11.3: sta: off, b: 100, h: 0, sa: 100, ct: 400; 11.4: sta: off, b: 54, h: 0, sa: 100, ct: 400
88580984 shelly_main.cpp:492     Up 88.56, HAP 0/1/16 ns 1, RAM: 204320/193860; st 0; 11.1: sta: off, b: 100, h: 0, sa: 100, ct: 400; 11.2: sta: off, b: 100, h: 0, sa: 100, ct: 400; 11.3: sta: off, b: 100, h: 0, sa: 100, ct: 400; 11.4: sta: off, b: 54, h: 0, sa: 100, ct: 400
96581490 shelly_main.cpp:492     Up 96.56, HAP 0/1/16 ns 1, RAM: 204320/193860; st 0; 11.1: sta: off, b: 100, h: 0, sa: 100, ct: 400; 11.2: sta: off, b: 100, h: 0, sa: 100, ct: 400; 11.3: sta: off, b: 100, h: 0, sa: 100, ct: 400; 11.4: sta: off, b: 54, h: 0, sa: 100, ct: 400
104581489 shelly_main.cpp:492     Up 104.56, HAP 0/1/16 ns 1, RAM: 204132/193860; st 0; 11.1: sta: off, b: 100, h: 0, sa: 100, ct: 400; 11.2: sta: off, b: 100, h: 0, sa: 100, ct: 400; 11.3: sta: off, b: 100, h: 0, sa: 100, ct: 400; 11.4: sta: off, b: 54, h: 0, sa: 100, ct: 400
112580945 shelly_main.cpp:492     Up 112.56, HAP 0/1/16 ns 1, RAM: 204132/193860; st 0; 11.1: sta: off, b: 100, h: 0, sa: 100, ct: 400; 11.2: sta: off, b: 100, h: 0, sa: 100, ct: 400; 11.3: sta: off, b: 100, h: 0, sa: 100, ct: 400; 11.4: sta: off, b: 54, h: 0, sa: 100, ct: 400

Log by activating channel 2: 

280423759 shelly_debug.cpp:231    No log file, sending new entries
[... cut ...]
301448782 shelly_output.cpp:96    Output 2: 0.005696 (transition)
301456736 shelly_light_bulb_co:56 Transition finished, end state: [w=0.01]
301460322 shelly_output.cpp:83    Duty: 0.000
304666602 mgos_sys_config.c:323   Saved to conf9.json
304690351 shelly_main.cpp:492     Up 304.67, HAP 0/1/16 ns 1, RAM: 203832/174652; st 0; 11.1: sta: off, b: 100, h: 0, sa: 100, ct: 400; 11.2: sta: off, b: 100, h: 0, sa: 100, ct: 400; 11.3: sta: off, b: 100, h: 0, sa: 100, ct: 400; 11.4: sta: off, b: 54, h: 0, sa: 100, ct: 400

markirb, 2025-01-20:
This sounds more like a PWM problem. 

Can you try to set the mode to normal (4 switches) on time and see if everything works there. 

Outputs are binary then.

stahlbox, 2025-01-20:
in switch mode it works fine... :)

I use 2 different LEDs for testing, it seems the second one has a problem with pwm. I swapped the LEDs (back in white mode), now channel 2 works and channel 1 is acting up.

Thanks for the solution - I'll test it again in a few days or next week with the same "correct" LED, but I'm confident that it will work then and will report back.

markirb, 2025-01-21:
Ok Strange. I remember there was someone having a similar problem that went away.

kaahla, 2025-01-21:
@stahlbox could you share your wiring setup please? I have issues with my setup:
I want to use the rgbw pm in white mode with two tunable white leds: channel 1+2 for led 1 and channel 3 + 4 for led 2. And I want to use two switches, one for led 1 and one for led 2. No matter how I connect the wires, I don’t get the second switch to work… don’t know what to select in the settings either :(
Thanks for your help!

stahlbox, 2025-01-22:
I use the circuit provided by Shelly with 2 LEDs and 2 buttons in white mode. Switched to channel 1 and 2. Channel 3 and 4 are not connected. Nothing special

kaahla, 2025-01-22:
@stahlbox where can I find this circuit? Cannot find it sadly, could you share it with me please? :-)

stahlbox, 2025-01-22:
[image]

FabrizioCivic, 2025-03-31:
Has anyone solved it? I tried with Wi-Fi saving both 0 and 1 or 2 but I didn't solve anything. When I turn on 1 channel, the other channels also turn on

stahlbox, 2025-03-31:
No solution. Same problem. Unfortunately. I'd like to use the shellys, but I can't.

markirb, 2025-04-01:
I will look into this. 

I fear this is a faulty mongoose implementation of pwm GPIO on esp32. 
This is the first device really using those on ESP32.

Will need to move to the esp native ones.

markirb, 2025-04-03:
I see the error now, it should be this: https://esp32.com/viewtopic.php?t=30862

I will try a fix today

markirb, 2025-04-03:
[shelly-homekit-ShellyPlusRGBWPM.zip](https://github.com/user-attachments/files/19591923/shelly-homekit-ShellyPlusRGBWPM.zip)

can you try this? this should fix it

FabrizioCivic, 2025-04-03:
Ok, I uploaded it, now it seems that the channels turn on separately, but if I try to turn them off, it turns off but turns on again immediately even if it is off and I have no way to turn off the channel

markirb, 2025-04-03:
I see. Almost there... I will build a new version

markirb, 2025-04-03:
new try: 

[shelly-homekit-ShellyPlusRGBWPM.zip](https://github.com/user-attachments/files/19593037/shelly-homekit-ShellyPlusRGBWPM.zip)

FabrizioCivic, 2025-04-03:
Ok, it seems that now it turns on and off correctly for every single channel, but when it turns off it makes a quick flash at maximum power and turns off

markirb, 2025-04-04:
[shelly-homekit-ShellyPlusRGBWPM.zip](https://github.com/user-attachments/files/19602793/shelly-homekit-ShellyPlusRGBWPM.zip)

ok this should fix that also

FabrizioCivic, 2025-04-04:
Yes! It works. I also flashed the other one and it works well too

markirb, 2025-04-04:
Great. Will be included in next release

closed 2025-04-10 by markirb