mongoose-os-apps/shelly-homekit #1617: 2.13.4-alpha2@2PMGen3: Calibration not working

hellspawn0815, 2025-01-21:
Hi there,

I am new to Shelly and Github, please be patient :) 

My Shelly2PMGen3 is flashed with the brand new v2.13.4-alpha2, build 20250121-122123. After updating and reseting I did the following:
- Giving a name and setting "Mode" to "Roller Shutter". Hitting save
- Setting "Input Mode" to "Separate - momentary" (That is what is have). Hitting save
- Hitting "Calibrate"
- Hearing the relay ticking, shutter drives upwards 
- Shutter reaching the top, shutter stops and is going silent. Nothing happens. Shutter is _not_ driving downwards again. Silence
- Roughly 10 Secs after the Shutter has reached the very top suddenly in the Shelly-WebUI directly under the "Shelly-HomeKit"-Logo the notification "Disconnected" appears.
- Still nothing happens. Shutter is _not_ driving downwards again. Silence
- After roughly 40 secs: relay is ticking again
- Still nothing happens. Shutter is _not_ driving downwards again. Silence
- Status of "Calibration" is still "not calibrated"
I was able to re-create this issue several times.

Any idea?
Anything I can help you with?

PS: When calibrating the shutter is _always_ driving up first, am I right? Has the shutter to be in a specific position before I start the calibration? Does it need to be at the very lower end? Or somewhere in the middle?

markirb, 2025-01-21:
Hi, 

Most likely that the power measurement channels are swapped in code.

Can you try to set your Shelly to normal mode. Then you see direct power readings per channel. If you switch on channel 1 you should read power on channel 1, and so on.

This would help to see if things are right or weong

hellspawn0815, 2025-01-21:
With "normal mode" you don't mean "Mode" = "Switch" or something? You likely mean shifting back to stock-firmware, don't you? Just quickly want to check before I do a re-config :)

If so: you just want me check in the stock app whether channel-1-switching appears in channel-1-power-readings, did I get that right?

PS: Prior to that I used beta firmware 1.5.0 and calibration went fine and the power consumption was displayed in realtime. And btw: When the shutter reached it's top position the power consumption dropped to 0 Watt

markirb, 2025-01-22:
I do mean switch mode yes.

See here for similar problem on first hw revision with screenshots of what to look for:

https://github.com/mongoose-os-apps/shelly-homekit/issues/1424

hellspawn0815, 2025-01-23:
Now I got you, thanks for clarification :)

I had a look a the other thread's screenshot. I am afraid on my WebGUI there is no power consumption being displayed. Also not when the shutter ist driving (Screenshot was taken while shutter was driving). 
[image]

First attempt was done by using Safari. Even with Firefox there is no power consumption being displayed
[image]

markirb, 2025-01-23:
Ah ok. Thank you for the information. This means that the PM chip could not be initialized. 

I will have to double check the code....

markirb, 2025-02-05:
Yes check the discussion thread. There is a version that has the PM. But not yet correctly scaled it is still off by a factor

hellspawn0815, 2025-02-05:
I guess you mean the [Shelly Homekit 2.14.0-alpha2](https://github.com/mongoose-os-apps/shelly-homekit/releases/tag/2.14.0-alpha2)-thread. But if I am switching to the code-section of [[Shelly Homekit 2.14.0-alpha2](https://github.com/mongoose-os-apps/shelly-homekit/releases/tag/2.14.0-alpha2) there is a roughly two week old download for [shelly-homekit-Shelly2PMGen3.zip](https://github.com/mongoose-os-apps/shelly-homekit/releases/download/2.14.0-alpha2/shelly-homekit-Shelly2PMGen3.zip). But that download contains the version "S2PMG3-2.**13**.4-alpha2" (what about 2.**14**?), but I have v2.13.4-alpha2 Build 20250121-122123 already installed (see above)

markirb, 2025-02-05:
I mean currently the version from here: (this is the most recent one) 
https://github.com/mongoose-os-apps/shelly-homekit/discussions/1585#discussioncomment-12033840

sorry for the confusion I will make a new alpha release once I get the scaling roughly right.

markirb, 2025-02-09:
Here is the new 2.14.0-alpha3: https://github.com/mongoose-os-apps/shelly-homekit/releases

This has Power Measurements. Those can be still be off by an integer factor, this has yet to be checked. Currently code matches +2PM

hellspawn0815, 2025-02-09:
Hey, just wanted to give feedback: According to your post 4 days ago I installed that [linked Firmware](https://github.com/mongoose-os-apps/shelly-homekit/discussions/1585#discussioncomment-12033840). And as of today I can say: Callibration is working 🥳 And as far as I can report as of today: Everything works fine 🥳

And thanks for the recent hint regarding [alpha3](https://github.com/mongoose-os-apps/shelly-homekit/releases). I installed that immediately and I am currently observing it's behavior and will report any unusual things 🫡

closed 2025-02-09 by hellspawn0815