mongoose-os-apps/shelly-homekit #1620: shelly 1pm gen1 reset (lose) settings - find ghost clicks in logs

lpt2007, 2025-01-31:
My shelly 1pm gen1 lose settings after couple of minutes.

here is log:
```
404546786 shelly_input.cpp:76     Input 0: change (state 0)
408773992 shelly_main.cpp:486     Up 408.76, HAP 0/1/12 ns 1, RAM: 25896/14616; st 54; 0.1: st:1 in_st:0 inm:3 ininv:0; 9.1: svt:9 inv:0 st:0 lea:-1.000
416774184 shelly_main.cpp:486     Up 416.76, HAP 0/1/12 ns 1, RAM: 26076/14616; st 54; 0.1: st:1 in_st:0 inm:3 ininv:0; 9.1: svt:9 inv:0 st:0 lea:-1.000
424774217 shelly_main.cpp:486     Up 424.76, HAP 0/1/12 ns 1, RAM: 26076/14616; st 54; 0.1: st:1 in_st:0 inm:3 ininv:0; 9.1: svt:9 inv:0 st:0 lea:-1.000
432774084 shelly_main.cpp:486     Up 432.76, HAP 0/1/12 ns 1, RAM: 26076/14616; st 54; 0.1: st:1 in_st:0 inm:3 ininv:0; 9.1: svt:9 inv:0 st:0 lea:-1.000
433476786 shelly_input.cpp:76     Input 0: change (state 1)
438681847 shelly_input.cpp:76     Input 0: change (state 0)
438687882 shelly_input.cpp:76     Input 0: single (state 0)
438692660 shelly_output.cpp:76    Output 1: on -> off (btn)
440977567 mgos_sys_config.c:323   Saved to conf9.json
440994737 shelly_main.cpp:486     Up 440.98, HAP 0/1/12 ns 1, RAM: 25868/14616; st 54; 0.1: st:0 in_st:0 inm:3 ininv:0; 9.1: svt:9 inv:0 st:0 lea:-1.000
441796864 shelly_input.cpp:76     Input 0: change (state 1)
448601880 shelly_input.cpp:76     Input 0: change (state 0)
448608003 shelly_input.cpp:76     Input 0: single (state 0)
448612773 shelly_output.cpp:76    Output 1: off -> on (btn)
448920836 mgos_sys_config.c:323   Saved to conf9.json
448938009 shelly_main.cpp:486     Up 448.92, HAP 0/1/12 ns 1, RAM: 25868/14616; st 54; 0.1: st:1 in_st:0 inm:3 ininv:0; 9.1: svt:9 inv:0 st:0 lea:-1.000
450036903 shelly_input.cpp:76     Input 0: change (state 1)
450356753 shelly_input.cpp:76     Input 0: change (state 0)
450544008 shelly_input.cpp:76     Input 0: single (state 0)
450549111 shelly_output.cpp:76    Output 1: on -> off (btn)
454016929 shelly_input.cpp:76     Input 0: change (state 1)
456920413 mgos_sys_config.c:323   Saved to conf9.json
456937584 shelly_main.cpp:486     Up 456.92, HAP 0/1/12 ns 1, RAM: 25832/14616; st 54; 0.1: st:0 in_st:0 inm:3 ininv:0; 9.1: svt:9 inv:0 st:0 lea:-1.000
463241980 shelly_input.cpp:76     Input 0: change (state 0)
463248098 shelly_input.cpp:76     Input 0: single (state 0)
463252868 shelly_output.cpp:76    Output 1: off -> on (btn)
464381923 shelly_input.cpp:76     Input 0: change (state 1)
464986612 mgos_sys_config.c:323   Saved to conf9.json
465003782 shelly_main.cpp:486     Up 464.99, HAP 0/1/12 ns 1, RAM: 25832/14616; st 54; 0.1: st:1 in_st:0 inm:3 ininv:0; 9.1: svt:9 inv:0 st:0 lea:-1.000
468031956 shelly_input.cpp:76     Input 0: change (state 0)
468038014 shelly_input.cpp:76     Input 0: single (state 0)
468042782 shelly_output.cpp:76    Output 1: on -> off (btn)
468251932 shelly_input.cpp:76     Input 0: change (state 1)
472921012 mgos_sys_config.c:323   Saved to conf9.json
472938187 shelly_main.cpp:486     Up 472.92, HAP 0/1/12 ns 1, RAM: 25828/14616; st 54; 0.1: st:0 in_st:0 inm:3 ininv:0; 9.1: svt:9 inv:0 st:0 lea:-1.000
478261365 shelly_input.cpp:76     Input 0: long (state 1)
478266435 shelly_reset.cpp:202    0: Reset sequence detected
478871588 shelly_reset.cpp:181    Performing reset
478876354 shelly_wifi_config_:156 Resetting STA settings
479115283 mgos_sys_config.c:323   Saved to conf9.json
479277490 mgos_sys_config.c:323   Saved to conf9.json
479290527 mgos_system.c:58        Rebooting in 2000 ms
479305120 shelly_sys_led_btn.c:44 === IDENTIFY ===
479311181 shelly_main.cpp:384     HAP server state: 2
479321854 HAPPlatformTCPStrea:303 HAPPlatformTCPStreamClose ts 0x3fff5a2c nc 0x3fff58ec 0/0/12
479332581 HAPPlatformTCPStrea:105 0x3fff402c 0.0.0.0:9002 HAP connection closed, ns 0/0/12 i 1
479342416 shelly_main.cpp:384     HAP server state: 0
479347262 shelly_main.cpp:357     === Destroying accessories
```

Probably those ghost clicks trigger reset?
```
478261365 shelly_input.cpp:76     Input 0: long (state 1)
478266435 shelly_reset.cpp:202    0: Reset sequence detected
478871588 shelly_reset.cpp:181    Performing reset
478876354 shelly_wifi_config_:156 Resetting STA settings
479115283 mgos_sys_config.c:323   Saved to conf9.json
479277490 mgos_sys_config.c:323   Saved to conf9.json
479290527 mgos_system.c:58        Rebooting in 2000 ms
```

Can I somehow disable input, because I don't use them?

markirb, 2025-02-03:
Currently there is no way to disable them.
This looks to me like a failing hardware. Did this problem occur all of a sudden?

markirb, 2025-02-06:
so this sounds like EMI. which Wifi Power Saving mode do you have? It could be better if you activate it...

lpt2007, 2025-02-09:
Sorry for delay:)

Where I can find Wifi Power Saving mode? How to activate it?

markirb, 2025-02-10:
https://github.com/mongoose-os-apps/shelly-homekit/wiki/WiFi-Settings