mongoose-os-apps/shelly-homekit #1624: 2.14.0-alpha3@2PMGen3: HomeKit: Wrong device type

hellspawn0815, 2025-02-11:
Hey Markus,

what is more easy for you? Do you want to have for ever finding a new issue or should I extend existing non-rated issues?

I defined the device as a roller shutter, not a garage door opener:
[image]

But in HomeKit it is displayed as garage door opener:
[image]

Not critical, but wanted to let you know.

Regards.

markirb, 2025-02-11:
Hi,

I doubt this is a real bug. This is caching of HomeKit.

It says this device is not responding...

 Br

hellspawn0815, 2025-02-12:
btw: That device was partially also reachable ;-) But don't now where this connection issues might come from, my Wifi works like a charm. Just fyi
[image]

Maybe it was related to the update to Alpha3. Anyhow, what I did today was to completely delete the device in HomeKit and resetting the HomeKit-settings on the Shelly itself and then re-paired it to HomeKit again. And now it is detected as a shutter 
[image]

Anyhow, it works now for me :) https://www.youtube.com/watch?v=t2F1rFmyQmY

closed 2025-02-12 by hellspawn0815