mongoose-os-apps/shelly-homekit #1626: Shelly very slow without HomeKit connection

zehrer, 2025-02-25:
I changed the WIFI network and now all my old devices are in a separate network (guest).
This affects as well a Shelly Plus 1 (which is in the wall).  Now the HomeKit host is not reachable anymore in this network.

Not when trying to access to the shelly it reacts very slow and it is not possible to update the network setting. 
Not clear what the device is doing but at least it is not possible anymore more control it.

zehrer, 2025-02-25:
[image]

This is the screenshot after trying to access it

matthias-zivim, 2025-02-25:
I had the same problem. The Shelly Plus 1 had been running for weeks (since v2.13.3 release in December, set up from 0) with WiFi 1 AP data setup. Then I thought about disconnecting my IoT from the “normal” WiFi.

I setup the ShellyPlus1 FW v2.13.3 with WiFi 2 SSID / password and then switched off WiFi 1. After that, the Shelly responded extremely poorly. Ping times went extremely high and the whole thing increased until the Shelly no longer responded to the button on the wall.

What I noticed was that although the Shelly was connected via WiFi 2 with the new SSID, the Shelly showed “ Status: Connected (reverted) ”

The only thing that helped was a software reset and setting up “WiFi 1” again (WiFi 2 is empty).

Now it works perfectly again.

zehrer, 2025-03-06:
I setup a temporary Wifi with the old setup to get access to. Afterwards I moved back to the original firmware which support bluetooth as well to access to device.

closed 2025-04-01 by markirb