mongoose-os-apps/shelly-homekit #1629: Shelly1 offline in Apple Home

stefanzilske, 2025-03-06:
I have several Shelly1 running on 2.13.3, that keep disconnecting from Apple Home. I can still reach the web interface and when I reboot from there, they are online in Apple Home again for some time.

Webinterface reacts pretty fast, all have static IPs assigned in my WiFi router. Any ideas for troubleshooting? I think this behavior started some time this year, can I downgrade the firmware?

markirb, 2025-03-14:
Without Logs and maybe a screenshot of the device I cannot tell you what's happening or change this.
Nothing regarding this behavior should have changed in the last versions.

djcargo, 2025-04-08:
I have the same problem with Shelly 1 Mini Gen 3 and Shelly Plus i4.

When Shelly disconnects from Homekit, it shows 0 pending and 0 active on web interface. And I can't display log in that situation.

I set a fixed IP and set power saving to disabled.

Earlier, when I was using Home Assistant to add this Shellys to HK, it wasn't disconnecting at all.

InventoryAndi, 2025-04-24:
Hi. I have the same issues with some of my Shellys. Here I have an example with my Bulb Duo "Tischlampe"

[image]

When I open it via ip in my browser I can see it directly and can turn on / off with no delays. In Apple Home sometimes there is no reaction at all "no answer" or long delays until on / off really appears. I use everywhere the latest firmware. I did not had this issue in the past. 1.13.XXI recently updated all my Shellys to the latest firmware and I had several with 1.13.XX firmware before.
I usually have for gen1 devices Power Saving 2, but I also tried it if disabled power saving. Same behavior. 

First I thought I have some wifi issues but other Shellys work at the same time great via HomeKit. So I just have this issues with some of them.

I don't know if logs will help you. Let me know if you need additional information. 

[image]
[image]

[Tischlampe.txt](https://github.com/user-attachments/files/19898209/Tischlampe.txt)

InventoryAndi, 2025-04-24:
Note: I realized this behavior especially with BulbDuo and RGBW (gen1). Maybe this helps...

InventoryAndi, 2025-05-16:
Hi. Any ideas / feedback here?

djcargo, 2025-05-16:
I have reverted firmware to stock and moved back to Home Assistant. And all works just fine.

For gen 3 devices there is new stock firmware which supports Matter. I will test it soon.

closed 2025-06-19 by github-actions[bot]