mongoose-os-apps/shelly-homekit #1630: Trying to flash 2PM GEN3 but nothing happens

Niflis, 2025-03-06:
Hey, i've been searching and trying to find others with similar issues on this device, but can't find any. 
I'm trying to flash my 2PM Gen3 using the http://192.168.x.xxx/ota?url=http://shelly.rojer.cloud/update URL, but it just says

"Trying to update from http://shelly.rojer.cloud/update" 
and nothing after that. The device isn't updating. 

I'm on the 1.4.99 firmware now, and can't find any new updates except for a 1.5.1 beta. 
Am i missing something obvious? or is there any settings i need to change on the device first for this to work? 

I've tried to reboot the device as well (via the UI, not cutting the power) 

can i flash it by downloading the firmware zip file somehow? the OTA option would be best for me ofc. 

Thanks in advance, and let me know if i missed something crucial info/logs etc to be able to solve this. 

all the best!

markirb, 2025-03-07:
This is still in alpha version, so you have to flash manually. This is: upload the zip via web if.

Please upgrade to latest version (in this case 1.5.1beta) before doing so.

Niflis, 2025-03-07:
Ah, ok. was thinking that maybe i could do that, but didn't want to do something wrong haha :) 
great, then i'll give it a go, I'll let you know if i notice any bugs/issues

Niflis, 2025-03-07:
It all worked fine after updating to the beta and uploading the zip to the shelly ui! 
Thanks for the help

markirb, 2025-03-15:
should work now, I made a release

closed 2025-03-15 by markirb

patrickweegen, 2025-03-30:
Should it work with the 1.5.1beta of the shelly firmware or also with the latest stable firmware?
I Installed today a new Shelly 2PM Gen3 but the script did not work

markirb, 2025-03-30:
What do you mean with the latest stable firmware? It does not work with 1.4.X

 They seem to be rolling out 1.5.1.

This firmware needs something bigger than 1.5.x. ....

patrickweegen, 2025-03-30:
I installed just another one. And Updated to firmware 1.5.1 befor trying to flash with the script. But it did not work.