mongoose-os-apps/shelly-homekit #1631: Shelly1 Fails to Reconnect via iPhone Home App When Using Wi-Fi Repeater

lukiiimohhU, 2025-03-13:
I have installed my Shelly1 device in a room located at a considerable distance from the main router. The Shelly1 is successfully connected to the Wi-Fi network, and I can confirm its connectivity as follows: it is visible in my router’s device list, I can ping it, and it is detectable using the Python tool provided in the associated repository.

When I am in the same room as the Shelly1 and connected to the Wi-Fi network, I can trigger my Home shortcut, and the Shelly1 performs its programmed actions without issue. However, a problem arises under the following circumstances: when I am elsewhere in my house, still connected to the same Wi-Fi network (using the same SSID and password) but via a different repeater located far from the main router, my iPhone fails to establish a connection with the Shelly1. In the iOS Home app, the device status displays "Reconnecting..." indefinitely, and it never successfully reconnects.

--------------------------------------------------------------

**Steps to Reproduce**

1. Connect the Shelly1 to the Wi-Fi network in a room distant from the main router.
2. Verify that the Shelly1 is connected and operational:
3. It appears in the router’s device list.
4. It responds to ping requests.
5. It is detectable using the Python tool provided in this repository.
6. In the same room as the Shelly1, connect an iPhone to the Wi-Fi network and trigger a Home shortcut. Confirm that the Shelly1 executes the programmed action successfully.
7. Move to another area of the house, still connected to the same Wi-Fi network (same SSID and password), but via a repeater located far from the main router.
8. Attempt to control the Shelly1 using the iOS Home app.

**Expected Behavior**
The Shelly1 should remain accessible and responsive through the iOS Home app, regardless of whether the iPhone is connected to the main router or the repeater, as long as it remains on the same Wi-Fi network.

**Actual Behavior**
When the iPhone is connected to the Wi-Fi network via the repeater (far from the main router), it fails to establish a connection with the Shelly1. In the iOS Home app, the device status displays "Reconnecting..." indefinitely and never successfully reconnects.

markirb, 2025-03-14:
Hi, first of all thanks for your detailed problem description.

As to why this only happens when you are far away I cannot really say.

Unfortunately I just think your WiFi is unreliable for the shelly or even more likely the **phone**. You may want to try if Stock Firmware behaves any different, but I doubt that it makes a difference.

"Reconnecting" only means that the webpage is loaded in your browser but the web socket connection got dropped.

lukiiimohhU, 2025-03-22:
Hello, I contacted Shelly Support regarding an issue that isn’t related to this repository. My Shelly device had an incorrect ID format, which prevented it from functioning properly over Wi-Fi. As a result, it was only operational via Bluetooth, which explains why it worked only when I was nearby.

Shelly Support has arranged a replacement for me, and I’ve just placed the order. I’ll give it another try once it arrives and update this thread if any issues arise. If you don’t hear anything further, assume everything worked out fine (just if someone sees this in the future).

markirb, 2025-03-23:
Ok I think I understand:

https://github.com/mongoose-os-apps/shelly-homekit/issues/1637

?

It is related to this app. We should fix it.

lukiiimohhU, 2025-03-23:
Yes, exactly the same problem as that issue.

Thanks for help @markirb, the problem should be on you side something with reformatting the correct IDs.

closed 2025-03-23 by lukiiimohhU