mongoose-os-apps/shelly-homekit #1632: Add Timed Status Logic for Garage Door Control without Sensors

jaap, 2025-03-14:
I’m using a Shelly 1 Plus to control my garage door, but I don't have the option to install a sensor to track the door's actual position. However, I know the timing of the garage door’s operation:

The door takes 10 seconds to open.

The door remains open for 30 seconds before automatically closing.

**Current Setup**:

I’m using the Shelly 1 Plus to close a circuit for 300ms, which acts like pressing a remote control button to operate the garage door. (I've removed the button and soldered 2 wires on it and put those on the IO points on the Shelly)

The garage door opens, but since no sensor is available, the state in HomeKit becomes inaccurate.

Would it be possible to introduce a feature that allows the Shelly 1 Plus to manage the state purely based on timing? For example:

Closed → Triggered

0 - 10 seconds → Opening

10 - 40 seconds → Open

40 - 50 seconds → Closing

After 50 seconds → Closed

This logic would ensure that HomeKit (or other systems) correctly reflect the door status without requiring physical sensors.

I believe this feature would be highly useful for users with similar limitations where sensor installation is impractical.

Since I'm really new to this Mongoose-project, I'm very open to diving in myself, but a bit of guidance from a more experienced contributor would very much be appreciated.

Thank you for your time and for the great work on this project!

markirb, 2025-03-14:
Hi,

This should be possible to implement it was requested before.

However I do not think it is very practical to do this with timing, only 30 seconds does not sound very common to me, what if your car or whatever is still in the way?

I just would make the closing sensor optional similar as the open sensor in the code can be optional. As the movement time is adjustable already this is then the just the timing base mechanism. 

Only problem is then to sync the state of the garage to the actual state of the door. If the Shelly uses power it comes up in a default state which would not match reality?

jaap, 2025-03-14:
I understand your concerns about relying purely on timing, but in my case, the precise state of the door isn’t crucial. The key point here is that the door always auto-closes after X seconds, so the timing logic would primarily serve to improve the HomeKit (or other system) status without relying on sensors.

The gate itself has a built-in safety mechanism that automatically reverses if it meets any resistance (like a car still in the way), so that’s not a concern. The timing-based approach isn’t meant to be perfect; rather, it’s a practical way to provide a reasonably accurate status without physical sensors.

Since this is a rolling gate that provides access to a shared lower-floor garage, I believe the simplified logic — transitioning briefly through “opening” before returning to “closed” — would be sufficient. This also makes it possible to immediately trigger the remote again if needed, which is crucial in case someone wants to quickly reopen the gate.

I fully understand that improving the sensor logic would be a cleaner solution in general, but adding an option for this simplified timing-based mechanism (with a configurable delay in the web interface) would be incredibly helpful for cases like mine where sensors aren’t feasible.

I’d be very grateful for any guidance you can provide on implementing this in the existing codebase — I’m eager to learn and contribute if needed.

Thanks for your time and insights!

markirb, 2025-03-14:
I just say timing will not be the best for everyone and thus merged to be in this firmware for everybody.

You can modify this firmware to fit your needs this is what open source is for. 

It is not easy to give guidance as I do not know about your background. Best to fire up GitHub codespace and talk with copilot to get going... also check the dev wiki.

jaap, 2025-03-16:
I modified the code, but could not get it on my Shelly unfortunately.
Then it daunted on me. All I actually need is a momentary switch. Turns out you can also choose the type of lock and combine that with an auto-off. This is enough for my use case. 

Instead of modifying the firmware, I just set the Shelly 1 Plus to act as a momentary switch with auto-off enabled. By setting the auto-off timer to 300 ms, it mimics a quick button press on the garage remote.

To make things even better, I marked the device as a lock in the settings. This closely resembles how a garage door behaves in the real world — you “unlock” it to trigger the remote, and after 300 ms, it automatically “locks” again, ready for the next press.

No complicated setup or sensor required, and it works reliably with my garage door’s built-in safety features. Simple and effective!

[image]

closed 2025-04-01 by markirb