mongoose-os-apps/shelly-homekit #1634: Total power consumption is not displayed  ShellyMiniPMGen3

tom666-max, 2025-03-16:
First of all, thank you for the extension.

Consumption is now displayed.
Total power consumption and costs are not displayed in the EVE app. 

Best Regards
Tom

[image]

markirb, 2025-03-17:
Hi, 

I guess you are looking for this pull request:
https://github.com/mongoose-os-apps/shelly-homekit/pull/1197

I rebased this onto current master now, in the pr action you can find a zip to test, can you try that?

tom666-max, 2025-03-17:
Hello Markus, 

thank you very much for your quick reply. There is no zip file for the ShellyMiniPMGen3 included in the PR action.

Kind regards
Thomas

markirb, 2025-03-17:
Are you seeing any? There are 30. amongst them is the one you are looking for.

tom666-max, 2025-03-17:
I only see this that I already have.  2.14.0 Releases 46

[image]

markirb, 2025-03-18:
It is also named 2.14.0 https://github.com/mongoose-os-apps/shelly-homekit/pull/1197/checks

you only see the matching commit id.

tom666-max, 2025-03-19:
Hello Markus, I don't quite understand. I've installed the latest firmware (2.14.0 Latest). When I access the Shelly directly, I can see the values. Not in the Eve app. I can't find any other zip file in the link either. I suspect I have to do something to expand the table in Homkit. So, what exactly do I need to do to display the total consumption and costs in Eve?

Kind Regards Thomas

markirb, 2025-03-22:
Hi, you upload this (right click save as)

[shelly-homekit-ShellyMini1PMGen3.zip](https://github.com/user-attachments/files/19401103/shelly-homekit-ShellyMini1PMGen3.zip)

Then you should see this 

[image]

and EVE should show precision of .1

[image]

Question is: should it be more precise then .1?
What should be the precision of Watt?

tom666-max, 2025-03-22:
Hello Markus

I installed this exact version and set everything up again.
Unfortunately, the data is still not being displayed.

Kind regards, Tom

[image]
[image]

markirb, 2025-03-22:
It is displayed with a precision of .1 everything underneath is neglected.Can you send a screenshot of your version and view in eve?Von meinem iPhone gesendet

Bildschirmfoto.2025-03-22.um.12.39.47.png (view on web)

Bildschirmfoto.2025-03-22.um.12.39.47.png (view on web)
Bildschirmfoto.2025-03-22.um.12.38.51.png (view on web)

tom666-max, 2025-03-22:
Hello Markus

Do you mean that?

Best regards, Tom

[image]
[image]
[image]

closed 2025-04-21 by github-actions[bot]