mongoose-os-apps/shelly-homekit #1637: Shelly Plus 1 PM - revert to stock firmware - Device ID issue

GEORGzer, 2025-03-18:
I have had a Shelly Plus 1PM for a while on stock firmware, so I could use the original Shelly app and the consumption diagrams and I used Homebridge for Homekit integration.

Today I played around a little bit and installed the Homekit Firmware, but then decided to go back to stock.

But then I suddenly had the issue that the Shelly wouldn't connect to the Shelly Cloud anymore. If I enable the Cloud, it briefly flashes an "connected at" address but that disappears immediately and when I refresh the page, Cloud is disabled again.

I noticed that the Device ID changed after reverting to stock firmware from what it was originally. Now it only shows the last 6 digits of the mac address, before it showed the whole address. I still have the old entry in the Shelly app to confirm that and I have another device from that generation on stock firmware that does it as well.

Tried to manually update to stock firmware from their servers and the beta, no change.

I believe either the Homekit firmware or the first revert process changed the device id and now I can't use this Shelly anymore with the Shelly Cloud on stock firmware.

markirb, 2025-03-18:
If you reset the device then anything we set is also reset, so if you want a clean slate then just do a real factory reset.

GEORGzer, 2025-03-19:
Thanks for the tip.
Factory reset fixed the issue, device ID is complete again and the Shelly was able to connect to the Cloud again 👍

markirb, 2025-03-23:
Want to keep this open until fixed. br

closed 2025-04-10 by markirb