mongoose-os-apps/shelly-homekit #1638: Shell +2PM reset after a power off

Florent003, 2025-03-22:
Hi all,

I received yesterday 8 Shelly +2PM for my covers.
I flashed all of em easily to the latest mongoose firmware. I can set up everything, it works fine.

But as soon as I power off my shellys (power off all my house for exemple) and then power back on, all of my shellys are reset.
They are still in Mongoose, but brand new flashed. Cover setup as gone, Wifi setup as gone, and they are in AP mode and accessible through 192.168.33.1.

I don't understand why.

If someone as the explanation, thanks a lot :)

Florent

markirb, 2025-03-22:
Might be that if you power up your house that your WiFi is also gone and then Shellys go to AP Mode? 

Are wifi setting the only thing gone or is everything gone?

Florent003, 2025-03-22:
All settings are gone, including cover settings.
If I switch off only the cover circuit breaker (so while my wifi is still up), I get the same result.

markirb, 2025-03-22:
Can you go to 

http://[deviceip]/rpc/Config.Get

and send me the value of "model"

Any specifics on your WiFi/Router Settings? Static IPs or anything?

Florent003, 2025-03-22:
Model is SNSW-102P16EU.
IPs are not set to static in the Shellys settings, but my Router always assign the same IP adress to the shellys based on their mac adress.

Florent003, 2025-03-22:
New tests done today : I downgraded the firmware to 2.13.3 and .... It's working fine.  No reset after a power off.

I conclude that there is an issue with firmware 2.14.0 and Shellys +2PM.

markirb, 2025-03-23:
Okay I see. Thanks for pointing that out.

Can it be that your device comes up in failsafe mode?
You show see no peripherals e.g a switch in WebIF?

shootsandroots, 2025-03-23:
Same thing just happened to me.

Florent003, 2025-03-23:
> e.g

There's no specificity with my Shellys. It's a standard mount.
As long as everything works fine with Shelly firmware and Mongoose 2.13.3, the issue seems very likely correlated to latest Moongoose 2.14.0.

markirb, 2025-03-23:
I reviewed the code two times now:

There is very slight changes that I had to make to Support also Gen3 but it should behave the same.

So it comes up with peripherals in the webapp? 
I tried to recreate this here but so far I could not

Florent003, 2025-03-23:
> So it comes up with peripherals in the webapp? I tried to recreate this here but so far I could not

I'm sorry I don't get your question

Marc-Azevedo, 2025-03-26:
I have the same problem

Florent003, 2025-03-26:
Try to downgrade your +2PM device to firmware 2.13.3.
I did it for all my +2PM Shelly last weekend and everything works like a charm even if there is a power cut.

Marc-Azevedo, 2025-03-26:
Yeah, that's what I did. Shelly 2.5 works fine with 2.14 but 2 PM doesn't

marmitol, 2025-03-26:
Same issue.
Yes, that's exactly what I did. The Shelly 2.5 works fine with firmware version 2.14, but the 2PM does not.
I downgraded the 2PM device to firmware version 2.13.3. I completely shut off the power to the entire building, and since then, the issue hasn’t reoccurred. Everything is working fine now.

markirb, 2025-03-26:
Okay, I mean it is definitely a confirmed bug, but currently not obvious in the diff from the two versions:

https://github.com/mongoose-os-apps/shelly-homekit/compare/2.13.3...2.14.0

I will have to try and recreate what's the issue here...

hughfr4nc15, 2025-04-11:
+1 with this issue and only on 2.14 firmware. I have some +2PM with 2.13.3 and they do not reset after power outage.

simonfield76, 2025-04-11:
Same issue. Following the thread

markirb, 2025-04-15:
Just to be sure: this does not  happen on 2.13.**3** ?

hananson, 2025-04-20:
Facing the same issue - after power down, two shelly 2PM+ loose entire config

pivale, 2025-04-29:
I have the same issue with +2pm, settings/definitions are reset after a power cut/outage.
I'll downgrade to 2.13.3 and will update on results

markirb, 2025-04-29:
[shelly-homekit-ShellyPlus2PM.zip](https://github.com/user-attachments/files/19965781/shelly-homekit-ShellyPlus2PM.zip)

can someone verify this fix please?

chris-rg, 2025-04-30:
Is it possible this happens to all Gen 3 devices? Same happened for me with Shelly 1 Mini Gen 3 and Shelly 1 Gen 3 with 2.14, but the second time it didn't even turn on AP mode again so I currently try to figure out how to reach them again.

markirb, 2025-04-30:
> Is it possible this happens to all Gen 3 devices? Same happened for me with Shelly 1 Mini Gen 3 and Shelly 1 Gen 3 with 2.14, but the second time it didn't even turn on AP mode again so I currently try to figure out how to reach them again.

No. +2PM is Gen2. The only reason why this happens to this device is because +2PM has dynamic pinning based on variant.

sello89, 2025-05-05:
I have one Plus2PM and have the issue with 2.14.
The last month i run with the 2.13.3 and have no power-off reset issue.
Yesterday i update to the new 2.14.2 and let the shelly from power over night.
The Reset happend again.

So back to 2.13.3 

Reset problem still there!

naanlizard, 2025-05-14:
Fwiw, I have four 2pm, only one of them reset when I had a big power outage, and that same one has been resetting seemingly randomly. I suppose it's possible there's a loose wire or something but I figured I'd report in just in case it helps

naanlizard, 2025-06-04:
Any progress here? Anything we can do to help?

raphtabe, 2025-06-05:
> Same issue with 5 device. Revert to 2.13.3 solve the issue. 4 devices cannot connect to my WiFi anymore. I’m still investigation. Does anyone else have the same issue ?

I solved the wifi connexion issue :

I reverted to stock firmware, connected via Bluetooth then configure the wifi, the changed the firmaware again

raphtabe, 2025-06-05:
> Any progress here? Anything we can do to help?

I’m stable with 2.14.2, but I don’t want to test a reboot, causing a lot of HomeKit reconfiguration in case of reset. 

Can I downgrade without loosing the HomeKit configuration ?