mongoose-os-apps/shelly-homekit #1641: Shelly rgbw white mode problem

FabrizioCivic, 2025-03-25:
Hi, I installed two rgbwpm Shelly, both set in white mode to power 2 strips on one while 3 strips on the other. The problem is that when I turn on a channel, even a second one turns on. To solve it I have to turn on all the channels of a Shelly and then turn it off and leave only the desired one active. This happens for both Shelly's. I've already installed the latest version and also tried the immediately previous one but nothing. Can anyone help me?
Thanks

FabrizioCivic, 2025-03-26:
Gen2. I must use the power saving in WiFi mode? 1 or 2 ?

markirb, 2025-03-27:
guess this is another issue duplicate of https://github.com/mongoose-os-apps/shelly-homekit/issues/1616

lets continue there and reopen this one.
I have yet to reproduce this effect.

closed 2025-03-27 by markirb