mongoose-os-apps/shelly-homekit #1646: ShellyPlusRGBWPM Firmware 2.14.0 crashes

tom666-max, 2025-03-29:
ShellyPlusRGBWPM firmware 2.14.0 crashes.
A revert or downgrade to 2.13.3 is no longer possible.

Regard Tom

markirb, 2025-03-29:
Hi,

Crashes how? 
Upgraded from 2.13.x?

There's absolutely no changes for this device in the firmware

tom666-max, 2025-03-30:
Hello Markus
Thank you very much for the quick reply.

Crashes: The connection to HomeKit simply drops. It remains accessible via the web GUI. Restarting the Shelly brings it back to live for a while until it shuts down again. All other Shellys (10x 1PM Mini Gen 3) are unaffected. To rule out the network, I restarted the router, and all Shellys except this one reconnected. I reset it and re-registered with HomeKit. Unfortunately, nothing changed.

Firmware Downgrade:  I reset the Shelly and tried to downgrade to 2.13.3 and the original 1.3.3 using the web GUI. In both cases, the process stopped after a short time. When I tried to use the original firmware, it briefly displayed the following message upon startup: 2.13.4 > 1.3.3.

I used the following stock firmware: https://rojer.me/files/shelly/stock/PlusRGBWPM.zip

Regard Tom

markirb, 2025-03-31:
This sounds more like a power dropout issue.

Try to disable ap and only have on source of WiFi. Try to set WiFi power saving to Maximum

tom666-max, 2025-04-01:
363765567 mgos_http_server.c:180  0x3ffcefb0 HTTP connection from 192.168.33.2:50464
364792305 mgos_ota_core.c:269     Starting, timeout 600, commit timeout 0, mem 200652
364812409 mgos_ota_core.c:656     FW: PlusRGBWPM esp32 1.3.3 20240625-122911/1.3.3-gbdfd9b3 sig 0x00
364818484 shelly_main.cpp:369     == Stopping HAP service (1)
364823223 mgos_ota_core.c:771     begin: waiting for user
364829098 shelly_main.cpp:384     HAP server state: 2
364838373 mgos_mongoose.c:66      New heap free LWM: 188848
364850128 HAPPlatformTCPStrea:112 0x3ffc0f68 0.0.0.0:9001 HAP connection closed, ns 0/0/16 i 1
364856780 shelly_main.cpp:384     HAP server state: 0
364861638 shelly_main.cpp:357     === Destroying accessories
368400231 shelly_main.cpp:492     Up 368.39, HAP 0/0/16 ns 0, RAM: 199676/188848; st 0; disabled
369791209 shelly_ota.cpp:90       Starting firmware update
369804613 esp32xx_ota_backend:209 App: PlusRGBWPM.bin -> app_1 enc 0, FS: fs.img -> fs_1 enc 0
369816260 mgos_ota_core.c:685     0.12% total, bootloader.bin 512 of 27440
370132425 mgos_ota_core.c:685     1.45% total, partition-table.bin 512 of 4096
370165636 mgos_ota_core.c:685     1.65% total, otadata.bin 512 of 8192
370736670 esp32xx_ota_backend:299 Skip writing app (digest matches)
370748128 mgos_ota_core.c:685     2.05% total, PlusRGBWPM.bin 512 of 1567040
373023871 mgos_ota_core.c:685     14.72% total, PlusRGBWPM.bin 262656 of 1567040
375206747 mgos_ota_core.c:685     27.39% total, PlusRGBWPM.bin 524800 of 1567040
376401868 shelly_main.cpp:492     Up 376.39, HAP 0/0/16 ns 0, RAM: 203752/188848; st 0; disabled
377188396 mgos_ota_core.c:685     40.07% total, PlusRGBWPM.bin 786944 of 1567040
379686571 mgos_ota_core.c:685     52.74% total, PlusRGBWPM.bin 1049088 of 1567040
382176553 mgos_ota_core.c:685     65.41% total, PlusRGBWPM.bin 1311232 of 1567040
383759426 mgos_ota_core.c:1051    Update finished, result -1 (File is too big)

tom666-max, 2025-04-02:
Thank you, that worked.
Thomas