mongoose-os-apps/shelly-homekit #1648: Help Flashing Shelly Plus2PM

rjbs1997, 2025-04-02:
Hey,
Im trying to go back to original firmware or mongoose os but im enable to do it.
Now, i have HAA and i wanna go to mongoose.
My ideas were to try to go back to shelly then to go to mongoose os.
I can flash to mongoose but seems like its bugged i'll explain: I can make up so shelly creates its own AP, i can connect to my WIFI but then shelly seems its 120º C and i cant add it to HomeKit.

What can i do?
I have shelly connect to pc via ftdi

Thanks in advance

markirb, 2025-04-03:
The problem is the following:
By flashing HAA you wiped the calibration partition.
This partition is especially important for +2PM since there is the information which version of +2PM you have (there are two with different pinnings).
So now you have the wrong pinning which leads to wrong temperature readout.

The Solution? Do you have a serial connection to your shelly? then you could flash the partition. 
Otherwise you can manually compile a version for the correct pinning.

rjbs1997, 2025-04-03:
Im kinda new to these things, can you help me? Im stuck right here and i cant move on

markirb, 2025-04-03:
You can only recover yourself from that situation.

Please do some research if you do not understand then use e.g chargpt, i have sketched possible solutions thats all I can do.

If you cannot apply then you’re stuck with HAA or tasmota. Also stock firmware will not work then.

rjbs1997, 2025-04-03:
Well, im not sure if it was the correct method, but i used other shelly plus2PM i got, its running already mongoose OS, i connected via serial, did a backup of the firmware then i flashed the backup on the shelly that wasn't working.
I know there are 3 known versions of shelly plus 2 pm and i cheched if they match, and for my luck they did.
Thanks for the help

markirb, 2025-04-03:
Okay yes thats it….
Maybe while you have this on serial can you get any clue for this: https://github.com/mongoose-os-apps/shelly-homekit/issues/1638

rjbs1997, 2025-04-03:
What can i do to help? I realized that the new version when you cut the power, the device forgets the wifi setting. All i did was install the previous version and now its working fine.

markirb, 2025-04-04:
Yes. Maybe after you cut the power you can capture a boot log and give me that so we can see what causes the reset?

Marfre888, 2025-04-22:
@rjbs1997 Any chance you can share that firmware dump? I'm stuck in the same hole and don't want to remove a shelly from my wall to do this. Shelly P2PM I have is version 0.1.9 (2022-8-2)

rjbs1997, 2025-04-22:
> [@rjbs1997](https://github.com/rjbs1997) Any chance you can share that firmware dump? I'm stuck in the same hole and don't want to remove a shelly from my wall to do this. Shelly P2PM I have is version 0.1.9 (2022-8-2)

There you go.
It is mongoose os

[backup_flash.zip](https://github.com/user-attachments/files/19855052/backup_flash.zip)

Marfre888, 2025-04-23:
> There you go. It is mongoose os
>
> [backup_flash.zip](https://github.com/user-attachments/files/19855052/backup_flash.zip)

Goes a lot quicker when you have the right tools in hand! Thank you so much for that, at least a few days of playing around with this wasn't for not. I owe you a beer!

closed 2025-04-29 by markirb