mongoose-os-apps/shelly-homekit #1651: Support for +uni analog input

shanse01, 2025-04-05:
Dear developers,

first of all a big thank you for the great work!

I came across your project when trying to implement a water level indicator for HomeKit with Shelly plus uni and its analog input. Works great, but thus far the analog inout is not supported by your firmware. Is there any chance that you might change this? HomeKit does seem to support a water level characteristic.

Kind regards
Sven

markirb, 2025-04-05:
I mean it would not be hard to read the analog value. 

But which water level property are you talking about? I did not find anything in the HAP Spec?

shanse01, 2025-04-05:
Hi Markus,
Thanks for the quick response.

Found this in the documentation: https://developer.apple.com/documentation/homekit/hmcharacteristictypewaterlevel

Please excuse as I am not a programmer, I might have gotten that wrong. If it needs to be an attribute of an accessory like a humidifier I could live with it as long as I can start an automation at or above/below a certain level in HomeKit.

HomeKit
HMCharacteristicTypeWaterLevel
Global Variable
HMCharacteristicTypeWaterLevel
The water level measured by an accessory.
iOS 10.2+
iPadOS 10.2+
Mac Catalyst 10.2+
tvOS 10.1+
visionOS 1.0+
watchOS 3.1.1+
let HMCharacteristicTypeWaterLevel: [String](https://developer.apple.com/documentation/Swift/String)
[Discussion](https://developer.apple.com/documentation/homekit/hmcharacteristictypewaterlevel#Discussion)
The corresponding value is a percentage of the maximum amount of water that the accessory, like a humidifier, can hold.

Again Thanks for the great work.

Regards
Sven

closed 2025-05-09 by github-actions[bot]