mongoose-os-apps/shelly-homekit #1664: Shelly2PmGen3 at Installation: Error fetching update: Bad HTTP response: 404

DaZim, 2025-05-01:
While installing with the Link http://A.B.C.D/ota?url=http://shelly.rojer.cloud/update on my Shelly the status in the Browser stays at "Trying to update from http://shelly.rojer.cloud/update" doesn't progress at all. 
On de Webinterface of the Shelly the Clear Console shows the following messages as shown in the picture below.
[image]

Shelly 2PM Gen3
Firmware ID: 20250318-152139/1.5.1-g01dd7ff

xkyle22x, 2025-05-02:
Mine is doing the exact same thing. 
Can't work it out just fails to update.

xkyle22x, 2025-05-02:
Updating via file worked. 
Thanks.

Maxym1979, 2025-05-03:
> Updating via file worked. Thanks.

Hi, what do you mean by "update via file"? I can't find a download for the Homkit firmware. I tried using a script, but I couldn't find the Shelly. Can you please help me?

xkyle22x, 2025-05-03:
> > Updating via file worked. Thanks.
> 
> Hi, what do you mean by "update via file"? I can't find a download for the Homkit firmware. I tried using a script, but I couldn't find the Shelly. Can you please help me?

Go to releases and download the zip file for your Shelly then go to setting of your Shelly in the web UI-firmware-update from file.

Maxym1979, 2025-05-03:
> Go to releases and download the zip file for your Shelly then go to setting of your Shelly in the web UI-firmware-update from file.

Thanks, found it and now all run,s.

vek030, 2025-05-10:
No install possible from Original Shelly 1.6.1 to HomeKit Version via File or OTA

markirb, 2025-05-10:
> No install possible from Original Shelly 1.6.1 to HomeKit Version via File or OTA

Not sure if this will be possible ever. 1.6.1 introduces matter which needs secure boot as far as I understand . Can you activate logs and  post an error message please.

DaZim, 2025-05-14:
> Sorry, no time to look into that yet. server seems to still be running so it might be a mapping problem. You can still use the manual link or update via file in the meantime.

That worked on my shelly 2pm gen3! Thank you!

markirb, 2025-05-22:
I fixed now the auto update. the mapping was wrong , but should be correct now

closed 2025-05-22 by markirb