mongoose-os-apps/shelly-homekit #1672: ShellyPlusPlugS Status LED can not be disabled

tilsee, 2025-05-11:
Hi,
After updating my ShellyPlusPlugs from  2.13.2 to 2.14.2 I can not deactivate the status led anymore.
I reverted to stock firmware and reinstalled mongoose which did not fix the problem.

When status led is activated, only half of the leds turn red or blue respective, while the other half stays red no matter what state the shelly is in.
When status led is deactivated all leds stay red in any shelly state.

If I can provide any additional Information I'll gladly provide it!

tilsee, 2025-05-26:
I rolled back to the latest 13.x Version for now which works perfectly, if one does not need the led indication.

markirb, 2025-05-26:
Sorry here's the zip:

[shelly-homekit-ShellyPlusPlugS.zip](https://github.com/user-attachments/files/20442144/shelly-homekit-ShellyPlusPlugS.zip)

JoJo03071967, 2025-05-26:
sorry, it doesn´t work the LED shows red all time

markirb, 2025-05-26:
ok. sorry, to be more precise: this device has four leds. the original issue description here was that only half of them turn the color.
is this the problem or do you want to be the light off when its off?
this can be adjusted by settings, which are not yet exposed to web frontend

JoJo03071967, 2025-05-26:
ah ok, I want to be the light off when its off? What can I do now?

JoJo03071967, 2025-05-26:
ok thank you, but I am not a expert in this case. How does it work? sorry for such a stupid question.

kenohassler, 2025-05-26:
Hi @markirb, thanks for the assistance! Using the firmware you provided, one red LED is still always on, no matter if you switch _State LED_ on or off.
I also tried the API commands you proposed:
```
$ curl http://192.168.188.38/rpc/Config.Set -d '{"config": {"led": {"color_off": 0x000000}}}'
{"saved": false}
$ curl http://192.168.188.38/rpc/Config.Set -d '{"config": {"led": {"color_on": 0x000000}}}'
{"saved": false}
```
I needed to reboot the plug for the changes to take effect, but now it's off. However, I went back to the current stable release (20250502-194309), and it behaves the same. So your change was probably not effective?