mongoose-os-apps/shelly-homekit #1681: Adding support with Shelly firmware 1.6.1

HugoTrick, 2025-05-25:
Hello,

I can't flash a shelly 2PM Gen3 from the last firmware version 1.6.1.

I think it's a new version and a problem of support.

Could you please add the support of this last Shelly firmware version ?

Thank you so much,
Hugo.

markirb, 2025-05-26:
> Hi Same problem here with Plus 1 PM, but the latest "revert" firmware is 1.4.4, which doesn't work. Any help greatly appreciated.

Can you please be more specific? 1.6.X -> 2.14.2 works for me on +1 PM.
Please enable logs in web and share any error message

markirb, 2025-05-26:
> I found a way to make it work.
> 
> If it can help anyone :
> 
> * Revert the firmware 1.6.X to 1.5.1 by using https://rojer.me/files/shelly/stock/latest/XXX
> * Than apply the procedure when you are in 1.5.1

okay I have looked into this. 
The problem is on 2PM G3 they changed partition layout again.
I have a build that changes accordingly, which again breaks this firmwares direct upgrade path....

It is getting tedious now to keep up with their changes.

Also Revert is broken to direct 1.6.X as they removed sha1 checksums from the zip bundle, we can repatch them afterwards then it works.... 
At least in general they seem to have found a solution to enable 3rd party firmware with matter, which is surprising

pmgilde, 2025-05-26:
Same problem here. 
I have a purchase of 10 units on hold waiting to see if you find a solution. 🙏🏻🙏🏻

markirb, 2025-05-27:
> > > I found a way to make it work.
> > > If it can help anyone :
> > > 
> > > * Revert the firmware 1.6.X to 1.5.1 by using https://rojer.me/files/shelly/stock/latest/XXX
> > > * Than apply the procedure when you are in 1.5.1
> > 
> > okay I have looked into this. The problem is on 2PM G3 they changed partition layout again. I have a build that changes accordingly, which again breaks this firmwares direct upgrade path....
> > 
> > It is getting tedious now to keep up with their changes.
> > 
> > Also Revert is broken to direct 1.6.X as they removed sha1 checksums from the zip bundle, we can repatch them afterwards then it works.... At least in general they seem to have found a solution to enable 3rd party firmware with matter, which 
> 
> > > Hi Same problem here with Plus 1 PM, but the latest "revert" firmware is 1.4.4, which doesn't work. Any help greatly appreciated.
> > 
> > Can you please be more specific? 1.6.X -> 2.14.2 works for me on +1 PM. Please enable logs in web and share any error message
> 
> Yes, 1.6.2 to 2.13.2. Can’t get logs at the moment, but after flashing it reports “successful”, but reboots to 1.6.2.  

Your problem on +1PM is very different from the one described here. 

Please see and continue here: https://github.com/mongoose-os-apps/shelly-homekit/issues/1547

pmgilde, 2025-05-27:
Hi markirb,

Please, can you explain how to make a downgrade on 2PM gen 3 to 1.5.x?

Thank you :)

markirb, 2025-05-27:
> Hi markirb,
> 
> Please, can you explain how to make a downgrade on 2PM gen 3 to 1.5.x?
> 
> Thank you :)

I would not do this, as in future we will support 1.6 only.

You can use this for now.
[shelly-homekit-Shelly2PMGen3.zip](https://github.com/user-attachments/files/20464784/shelly-homekit-Shelly2PMGen3.zip)

As said this does not allow revert currently, I have a solution but it has yet to be deployed.

pmgilde, 2025-06-07:
Hi!!!  

Any new update?
I have to buy a lot of units to control the blinds in my new house, but I didn't want to move forward without seeing this problem resolved.

Thank you so much!
Pedro

markirb, 2025-07-14:
it is almost done. I think latest next week but do not quote me on that.

pmgilde, 2025-07-21:
Hi Markirb!!! Just a kindly reminder!!! 😜

markirb, 2025-07-23:
should be possible now.

closed 2025-07-23 by markirb

pmgilde, 2025-07-23:
> should be possible now.

tested and worked perfectly.

Tank you so much!! 😊