mongoose-os-apps/shelly-homekit #1684: Doorbell Shelly Plus Uni

nqrlz, 2025-06-12:
I have installed a doorbell system in my house. The doorbell wires are connected exclusively to one of the inputs on the Shelly Plus Uni. One of the outputs is connected to the door opener wires. The setup works flawlessly with the Shelly software: when the doorbell is pressed, I receive a push notification on my phone, and I can open the door using the output switch.

However, when I install Shelly-HomeKit, I can no longer see when the doorbell is pressed. I’ve already tried several options. Here is my latest configuration status.

Any help is welcome—thank you very much!

[image]

markirb, 2025-07-10:
I think the bug is that the input pins currently used is not correct, they were taken from another template, no Idea why they would be wrong.

[fw.zip](https://github.com/user-attachments/files/21154227/fw.zip)
You can use this firmware afterwards you can probe for the pins via RPC like:

curl [ip of device]/rpc/GPIO.Read -d '{"pin":3}'

the correct ones have to change from 0 to 1 and vice versa when input is changed

janernsting, 2025-07-10:
> unfortunately my knowledge is not sufficient to follow you. can i fix the problem with the firmware that markirb has provided?

No, the firmware won't fix it. I assume markirb provided it for investigating the bug.

janernsting, 2025-07-10:
Weirdly enough, the pins are [mapped correctly](https://github.com/mongoose-os-apps/shelly-homekit/commit/733a97d40c659fca6950949f2fd36858ded60259#diff-2f35920310afc68ce023bce0cd046607f0cf8ceded129b92bec01d7c19f360c5R1257). Could it have to do with the way sensors are wired to the Plus Uni?

On the Shelly Plus Uni, IN1 and IN2 are connected to ground. On the Shelly Uni, IN1 and IN2 connect to L(ive). See wiring schematics below.

[image]

[image]

markirb, 2025-07-10:
> > unfortunately my knowledge is not sufficient to follow you. can i fix the problem with the firmware that markirb has provided?
> 
> No, the firmware won't fix it. I assume markirb provided it for investigating the bug.

Yes exactly.
I looked at it and I am sure the pins used currently are not correct as no such pins exist on the package.

markirb, 2025-07-12:
can you try the build I sent maybe it indeed works because it does not use noisy input pin, which might be unsupported on ESP32 somehow

janernsting, 2025-07-12:
> what voltage are you running this on?

28 VAC

> can you try the build I sent maybe it indeed works because it does not use noisy input pin, which might be unsupported on ESP32 somehow

Darn. The build you sent works 😀👍

While you are at it. On both Uni models (the non-plus and the plus variants) outputs are swapped. Hardware out 1 corresponds to HAP switch 2 and vice versa.

markirb, 2025-07-13:
Okay. Indeed it seems the noisy pin mode which causes trouble. I do not know why this should work in PlusI4 which is also ESP32...

Anyway, I will upload a fixed version here, and then the next release will contain that.

(on Mac: right click save as to download file without extracting...)
[shelly-homekit-ShellyPlusUni.zip](https://github.com/user-attachments/files/21204182/shelly-homekit-ShellyPlusUni.zip)

markirb, 2025-07-13:
Second correction now also with swapped outputs (matching the HW description )

nqrlz, 2025-07-15:
> [shelly-homekit-ShellyPlusUni.zip](https://github.com/user-attachments/files/21204476/shelly-homekit-ShellyPlusUni.zip) (on Mac: right click save as to download file without extracting...) Second correction now also with swapped outputs (matching the HW description )

thank you so much!

i'm trying to install the firmware via the webinterface, but it does not work. 

i add the .zip to the "Firmware file image" window, hit the "Update from file" button, get a small loading bar at the top of the browser, nothing happens.

what am i doing wrong :(?

closed 2025-07-23 by markirb