mongoose-os-apps/shelly-homekit #1690: GUI says “Disconnected” and loading after update

jezdinskyjiri, 2025-07-10:
Hi,
I updated to the latest fw, however it seems something went wrong for all my shellys. 

[image]

Any clue what I need to do to fix it?

markirb, 2025-07-12:
Please first describe your issue more detailed. 

What model?
What version upgrade to what version?

Can be that the device is simply in AP mode...

pfrasun, 2025-07-12:
I have exactly same issue with Shelly i4 DC with 2.14.2.
It works sometime when I use a sting xxx.xxx.xxx.xxx/#reboot, but it is inconsistent and unstable.
It worked perfectly for good year, but stopped recently.
After going back to 2.14.1 problem seems to still exist. It is not a browser issue as I tried after removing history etc. I will try reset and see if the issue gets solved.

pfrasun, 2025-07-12:
Additional Information: In my case, it must be related to the Safari web browser in iPadOS 26 Beta. All other devices work properly and also opening the interface using the Host address in iPad is working. I assume it has nothing to do with FW.

jezdinskyjiri, 2025-07-12:
Thanks everyone and sorry for little info. Ae mentioned by @pfrasun the issue is iOS 26, I have no issues accessing it on other devices not running 26.

markirb, 2025-07-13:
Ok then it is most likely related to http requests.... let's see if this problem resolves itself

closed 2025-08-12 by github-actions[bot]

pnuding, 2025-11-22:
Another curious finding: The page **does** load if I do the following:

1. Open the device's address in a private window 
2. in the hamburger menu on the left of the address bar go to website settings
3. enable or disable content blockers (doesn't matter which way)
(it doesn't seem to be related to the content blockers as such, because it doesn't matter if you enable or disable them, and "reload without content blockers" doesn't have the same effect)

or 
1. from the develop menu select "Empty Caches"
2. reload the page
(it only works **once** for each device, if you reload the page it's broken again until you empty the caches again. Specifically clearing the website data for the device from Settings>Privacy>Manage Website Data does **not** have the same effect. It looks like something specific to the device is cached and causes problems.)

pnuding, 2026-01-21:
just suffered this again yesterday, it doesn't seem like there's interest in fixing this so for now I've downgraded all devices to 2.14.1 (via the upload file button) and will see if this improves the situation. Has anyone else tried downgrading?

Cibernello, 2026-02-14:
Is anyone still experiencing this issue after upgrading to iOS 26.3? Thanks

pnuding, 2026-02-15:
Yes, it still happens to me on 26.3 ☹️