mongoose-os-apps/shelly-homekit #1696: HomeKit device type configuration not persistent on Shelly Plus 1

amrumer, 2025-08-08:
## Device Information
- **Model:** Shelly Plus 1
- **Firmware Version:** 2.14.3 (Shelly-HomeKit)
- **HomeKit Version:** iOS 18.6
- **HomeKit App:** Native iOS Home app

## Problem Description
The Shelly Plus 1 is configured as a Light accessory in HomeKit, but this configuration is not persistent. The device randomly reverts to being displayed as a Switch, and sometimes even appears as two separate Switch accessories overnight or after some time.

## Expected Behavior
The device should maintain its configured accessory type (Light) consistently and not change to Switch or duplicate accessories.

## Actual Behavior
- Device is initially configured as Light in HomeKit
- Overnight or after some time, it reverts to Switch accessory
- Sometimes appears as two separate Switch accessories instead of one Light
- This happens repeatedly and unpredictably

## Additional Information
- No error messages are displayed
- Issue occurs repeatedly
- Device functionality appears to work, only the accessory type changes
- Happens specifically overnight/after extended periods

Has anyone else experienced this issue with Shelly Plus 1 devices? Any known workarounds or fixes?

lalolerry, 2025-08-20:
I had experienced this strange behaviour!
I had a Shelly Plus 1 configured as outlet in Homekit App that worked seamless  with firmware 2.12. 
till I decided to upgrade to latest Mongoose version (that meant to go back to stock firmware, update to v.1.6 stock, than upload via OT fw 2.14.3)... and this same strange behavior appeared (in Homekit app sometime randomly it reverts to switch accessory, sometimes appears as double switch -one not working-, sometime lost accessory configuration -icon, show on favourites, postiion, etc...). 
After some time I decided to cancel it from homekit app and than readd... and since than this strange behaviour never showed up again.
I think this is related to a partially lost of homekit configuration in the tricky "upgrading" process.
Try it and let us know ig I'm right
Best Regards

amrumer, 2025-09-03:
Hi,

Unfortunately, even my second trial didn’t work. Here’s what I’ve done:

- I deleted the device from the Apple Home App.
- I reset the Shelly to its original firmware version.
- I updated the Shelly software to the latest version.
- I reset the device and renamed it.
- I flashed the Shelly to the Mongoose firmware.
- I set up the Shelly in the Apple Home App as a new device with a new name.

Initially, everything worked fine, but after two or three days, my problem occurred again. Now, when I check the Shelly in the Apple Home App, it’s no longer a bulb but a switch. 

Any further ideas or suggestions?