mongoose-os-apps/shelly-homekit #1712: Support for Ogemray 25A Relay?

dwiller11, 2025-09-19:
Any chance this could work with the Shelly Ogemray relay? Would be awesome if so.

markirb, 2025-09-19:
This seems to be a ShellyX attached to some relay, that's why its called "powered by shelly".

From the pictures in 
https://shelly-forum.com/thread/28789-teardown-ogemray-25a-smart-relay-powered-by-shelly/
we can see a lot of information:

- It is an ESP32C3 Chip so generally it would be no problem to support it.
- The exact shellyX model seems not to be documented on the shellyx website: https://x.shelly.com/modules/
- The Powermeasurement seems to be done via UART, however I cannot recognize the Chip in the pictures so no way of knowing if it is supported yet
- The Pinning is unknown, However the PCB Silkscreen has quite some information. Only thing unknown is the 6 IO numbers (LED, Resetbutton, S1 Input, SY Output, UARTRX/TX), that are printed on the side of the shellyX that is facing inward.

So summary:
I think this is very much possible
If you have a device and can give me the information then just supporting the switching mechanism is not a big deal.
Powermeasurement is unclear if it is easy or more work to support an unknown chip.

BEskandari, 2025-11-16:
> * The Pinning is unknown, However the PCB Silkscreen has quite some information. Only thing unknown is the 6 IO numbers (LED, Resetbutton, S1 Input, SY Output, UARTRX/TX), that are printed on the side of the shellyX that is facing inward.
> 
The photos of the other side of ShellyX. Hope that this could help:
[https://shelly-forum.com/thread/28789-teardown-ogemray-25a-smart-relay-powered-by-shelly/?postID=326674#post326674](url)

Looking forward to see if we can have the homekit integration.

markirb, 2025-11-17:
[image]
I can only read parts of that:
I would say:
IO3, IO4, GND, 3V3, IO????, IO6, IO7, --, IO5, IO10 ?

BEskandari, 2025-11-21:
Here is a clear view of the IO ports.
[https://shelly-forum.com/thread/28789-teardown-ogemray-25a-smart-relay-powered-by-shelly/?postID=327744#post327744](https://shelly-forum.com/thread/28789-teardown-ogemray-25a-smart-relay-powered-by-shelly/?postID=327744#post327744)
 IO3, IO4, GND, 3V3, ADC, IO6, IO7, --, IO5, IO10
Hope that this helps to have a Homekit enabled firmware.